makerbase-mks/MKS-THR36-THR42-UTC #3: Firmware in MKS UTC V1

acguardia, 2023-01-26:
I can't upgrade firmware in my MKS UTC V1.
The DFU-Upload-firmware.bat doesn't work
---------------------------------------------------
D:\MKS\CanBus\MKS-THR36-THR42-UTC-main\MKS UTC DFU-Upload\UTC_DFU_Upload>dfu-util.exe -a 0 -s 0x08000000:leave -D klipper.bin -d 0483:df11
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
No DFU capable USB device available

sephthir, 2023-02-19:
I am also running into this. Running `dfu-util -l` should show the device after plugging it in with the `BOOT` button held, but it returns that no DFU-capable device is present.

terraphantm, 2023-02-27:
The issue is makerbase configured the boot bits wrong. You can fix it with an STLink. You'll need to set the connection mode to hardware_reset to successfully connect. Then you need to set nBOOT_sel to 0. Afterwards the boot button will work properly.

NexGen-3D-Printing, 2023-04-03:
> The issue is makerbase configured the boot bits wrong. You can fix it with an STLink. You'll need to set the connection mode to hardware_reset to successfully connect. Then you need to set nBOOT_sel to 0. Afterwards the boot button will work properly.

Incorrect, what you do is put it back in the packet and send the piece of crap back for a refund, it should work out of the box, if they provide updated firmware it should work the way they state, if it doesn't, then its DOA, we should not be beta testers or expected to perform complex tasks like learning how to use an STLink so we can repair their dodgy inept work.

SnakeKVC, 2023-07-13:
not active
[image]

SnakeKVC, 2023-07-13:
[image]
Enable DFU with stm32CubeProgrammer.
Work for me.

ueatinmycookie, 2023-07-14:
Is there a certain driver for windows 10 that I may be missing? I can't connect to the board via stlink and it shows up in device manager under other devices.

ueatinmycookie, 2023-07-14:
Thanks. I still have an issue connecting to stlink. Error: No debug probe detected.

SnakeKVC, 2023-07-14:
> Thanks. I still have an issue connecting to stlink. Error: No debug probe detected.

have a screenshot with error?
My ST-Link v2
[image]

ueatinmycookie, 2023-07-14:
I have a clone of that one. Are the swd pin numbers on the utc board the same as the stlink?

SnakeKVC, 2023-07-14:
> I have a clone of that one. Are the swd pin numbers on the utc board the same as the stlink?

Only four wire need
[image]

Korgoth1234, 2023-07-20:
I tried to do as you described, but I get an error when I try to set nBOOT_SEL to 0:
```
Error: flash loader cannot be loaded.
Error: Downloading Option Bytes Data failed
```
[image]

SnakeKVC, 2023-07-20:
Wrong checkbox removed
[image]

Korgoth1234, 2023-07-20:
> Wrong checkbox removed Swap this [image]
No, I did everything right, you can see that from the log window:
[image]

SnakeKVC, 2023-07-20:
Why did you attach the wrong screenshot?

Korgoth1234, 2023-07-20:
> Why did you attach the wrong screenshot?
So this shows the stm32cube after I clicked "apply".

Korgoth1234, 2023-07-20:
I recorded a video of my actions:
https://github.com/makerbase-mks/MKS-THR36-THR42-UTC/assets/16425273/66b319ac-8299-4d82-9125-7db35a010223

SnakeKVC, 2023-07-20:
On my screenshot with chip stm32g0b1 don't have "DUAL_BANK" options.
I don't undestand.