makerbase-mks/MKS-THR36-THR42-UTC #8: Thermistor wrong readings THR36

Asica10, 2023-02-19:
Problem with temperature reading on THR36.
If I use the same thermistor and connect to Monster8 v2  instead CAN board THR36 reading are valid.
Difference between reading by THR36 and Monster8 is more than 5 degree.   In both case sensor is defined as Generic 3950. 
Seems to be 2 option - problem with board or  pull up resistor for TH id different than default 4.7k ohm. 
Any one have the same issue?

tome9111991, 2023-02-20:
i have same issue, had use a Generic 3950 and ATC semitec.
both with 4-5 degree lower as with my ebb36

tome9111991, 2023-02-21:
I have connected a Geniric 3950 to Octopus board:
[image]

and here same sensor connected to Thr36
[image]

Asica10, 2023-02-21:
I did additional measurements using resistor instead ntc100k to simulate the same condition:

Monster8v2
4.67k ohm - 113.3 C
1082ohm - 170.8 C
463ohm - 211.2 C 
195ohm - 258.8 C

THR36
4.67k ohm - 112.2 C
1082ohm - 169.4 C
463ohm - 207.4 C 
195ohm - 250.5 C

Difference between these 2 boards are not constans

SystemConfirmedInitiate, 2023-03-22:
I have the same issue as well!!

MKS-hosiey, 2023-03-23:
> I have the same issue as well!!

[mcu MKS_THR]
serial: /dev/serial/by-id/usb-Klipper_rp2040_0599429432067358-if00

#####################################################################
#              fan
#####################################################################

## FAN0  
[heater_fan hotend_fan]
pin:MKS_THR:gpio1
max_power: 1.0
kick_start_time: 0.5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

## FAN1  
[fan_generic fan0]
pin: MKS_THR:gpio2
max_power: 1.0
kick_start_time: 0.5
off_below: 0.10

[fan_generic fan1]
pin:MKS_THR:gpio3
max_power: 1.0
kick_start_time: 0.5
off_below: 0.10

#####################################################################
#   Enstop
#####################################################################

[stepper_x]
endstop_pin: MKS_THR:gpio24

[stepper_y]
endstop_pin: MKS_THR:gpio25
#####################################################################
#   Extruder
#####################################################################
[adc_temperature extruder]
temperature1:27
voltage1:3.156
temperature2:34.8
voltage2:3.10
temperature3:42
voltage3:3.01
temperature4:81.0
voltage4:2.354
temperature5:140.7
voltage5:1.029
temperature6:171
voltage6:0.619
temperature7:234
voltage7:0.222
temperature8:300
voltage8:0.098
[extruder]
step_pin: MKS_THR:gpio5
dir_pin: MKS_THR:gpio4
enable_pin: !MKS_THR:gpio10
rotation_distance: 22.4  #22.6789511	#Bondtech 5mm Drive Gears
gear_ratio: 50:17				
microsteps: 16
full_steps_per_rotation: 200	#200 for 1.8 degree, 400 for 0.9 degree
nozzle_diameter: 0.400
filament_diameter: 1.75
heater_pin:MKS_THR:gpio0
sensor_pin: MKS_THR:gpio26
sensor_type:extruder
adc_voltage: 3.3
voltage_offset: 0.00
#   The ADC voltage offset (in Volts). The default is 0.
#pullup_resistor:4700
#inline_resistor:0
min_temp:-270
max_temp: 380
max_power: 0.6
min_extrude_temp: 170
control = pid
pid_kp = 26.213
pid_ki = 1.304
pid_kd = 131.721
pressure_advance: 0.03
pressure_advance_smooth_time: 0.040

[tmc2208 extruder]
uart_pin: MKS_THR:gpio6
interpolate: false
run_current: 0.5
hold_current: 0.2
sense_resistor: 0.110
stealthchop_threshold: 0

#####################################################################
#	  Neopixel
#####################################################################

#[neopixel my_neopixel]
#pin: MKS_THR:gpio20
#chain_count: 10
#color_order: GRB
#initial_RED: 0.0
#initial_GREEN: 0.0
#initial_BLUE: 0.0

#####################################################################
#	ADXL345
#####################################################################

[adxl345]
cs_pin: MKS_THR:gpio13
spi_software_sclk_pin: MKS_THR:gpio14
spi_software_mosi_pin: MKS_THR:gpio15
spi_software_miso_pin: MKS_THR:gpio12

#####################################################################
#	Probe
#####################################################################

#[probe]
#pin: ^MKS_THR:gpio21
#x_offset: 0
#y_offset: 19.75
#z_offset: 7.45     #6.25
#speed: 7
#samples: 3
#samples_result: median
#sample_retract_dist: 0.5
#samples_tolerance: 0.05
#samples_tolerance_retries: 3

#[bltouch]
#sensor_pin:^MKS_THR:gpio21
#control_pin:MKS_THR:gpio11
#x_offset: -44
#y_offset: -14
#z_offset: 0.30
#speed: 15.0
#samples: 5
#samples_result: median
#sample_retract_dist: 3.0
#samples_tolerance: 0.10
#samples_tolerance_retries: 1

MKS-hosiey, 2023-03-23:
> I have the same issue as well!!
You can use the above configuration, which is a temperature correction configuration

SystemConfirmedInitiate, 2023-03-23:
I have used that configuration of course, gives me a reading 4 degrees off.

Asica10, 2023-03-23:
I had also deviation like you and prepare some adjustment and is much better. You can try below config:

[adc_temperature extruder]
temperature1:27
voltage1:3.156
temperature2:34.8
voltage2:3.10
temperature3:42
voltage3:3.01
temperature4:81.0
voltage4:2.354
temperature5:143
voltage5:1.029
temperature6:171
voltage6:0.632
temperature7:195
voltage7:0.401
temperature8:225
voltage8:0.250
temperature9:255
voltage9:0.159
temperature10:287
voltage10:0.098

tome9111991, 2023-03-23:
> No such luck unfortunately, that didn't work. 4 degrees off still.

you have change to sensor_type:extruder ?

NexGen-3D-Printing, 2023-04-15:
I have this same issue with a PT1000, exactly 5 degrees off at room temp

mapinto-3d, 2023-05-21:
Anyone has an update on this problem? I'm having the same issue here an no clue how to solve it.

Asica10, 2023-05-22:
Partialy you can solve problem by defining [adc_temperature extruder] and adjust to you. 
Unfortunately support from manufacturer is really bad...

kevinkga, 2023-07-20:
I'm confused. Does this support PT1000 at all? AFAIK chips like MAX31865 do but in this case I couldn't even find such a chip on the board. Does anyone know what chip it uses and if this does actually support PT1000?

MKS-hosiey, 2023-07-20:
> I'm confused. Does this support PT1000 at all? AFAIK chips like MAX31865 do but in this case I couldn't even find such a chip on the board. Does anyone know what chip it uses and if this does actually support PT1000?
> 
> Could this be related to [Klipper3d/klipper#4859](https://github.com/Klipper3d/klipper/pull/4859) ?

It supports PT1000, pt1000 does not need max31865, it can be directly connected to the thermal interface, and then configure pt1000

kevinkga, 2023-07-20:
> It supports PT1000, pt1000 does not need max31865, it can be directly connected to the thermal interface, and then configure pt1000

So you're saying I can connect a PT1000 sensor to the board, without an amplifier like the MAX31865? Are you saying that the [adc_temperature extruder] will replace the role of the MAX31865 chip in this case?

I'm trying to understand how this works

kevinkga, 2023-07-20:
> It supports PT1000, pt1000 does not need max31865, it can be directly connected to the thermal interface, and then configure pt1000

Thanks for the update. That helped clarify a LOT. 

So finally, does the board have a 4.7k pull-up resistor or a 1k resistor?