makerbase-mks/MKS-THR36-THR42-UTC #10: Pin 'gpio6' is not a valid pin name on mcu 'MKS_THR' in THR36

tntclaus, 2023-07-01:
I flashed firmware to THR36  as it is described in instruction: 

- mounted rp2040 as USB drive and copied uf2 file there
- after executing `sudo sync` rp2040 USB drive automatically  disconnected 

`~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0` command returns  uuid, which I use in `mcu` config section. The config is sample config from this repo. I didn't change anything. 

But it seems like firmware update actually doesn't happen, here is klippy log related to it trying to talk to THR36:

```
mcu 'MKS_THR': Starting CAN connect
Created a socket
mcu 'MKS_THR': Unable to open CAN port: Failed to transmit: [Errno 100] Network is down
Created a socket
Loaded MCU 'MKS_THR' 104 commands (v0.10.0-530-g3387a9c2-dirty-20221025_234520-mkspi / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-12+11) 2.31.1)
MCU 'MKS_THR' config: BUS_PINS_i2c1=PB6,PB7 RECEIVE_WINDOW=192 BUS_PINS_spi2a=PC2,PC3,PB10 BUS_PINS_i2c1a=PF1,PF0 RESERVE_PINS_CAN=PB8,PB9 STEPPER_BOTH_EDGE=1 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 CLOCK_FREQ=64000000 BUS_PINS_spi1=PA6,PA7,PA5 RESERVE_PINS_USB=PA11,PA12 MCU=stm32g0b1xx STATS_SUMSQ_BASE=256 BUS_PINS_spi1a=PB4,PB5,PB3 BUS_PINS_spi2=PB14,PB15,PB13
```

_This line looks wrong_: MCU 'MKS_THR' 104 commands (**v0.10.0-530-g3387a9c2-dirty-20221025_234520-mkspi** [...]  — my current git repo is after tag v0.11.0

_Also this looks weird_: MCU=**stm32g0b1xx** — WHY `stm32`? It should be `rp2040`. The chip of THR36 is indeed rp2040.

Finally, pin names are not of `rp2040` too. 

This is last part with error:

```
Sending MCU 'MKS_THR' printer configuration...
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 180, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 753, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 720, in _send_config
    % (enum_value, self._name))
Error: Pin 'gpio6' is not a valid pin name on mcu 'MKS_THR'
```

WTF? :) WHAT AM I DOING WRONG? How is this even possible?

tntclaus, 2023-07-01:
Idk why, but CAN is not working on my board. Flashing with firmware that should communicate via USB works. Flashing it with CAN firmware doesn't work.

Armadillo1, 2024-03-01:
I had exactly the same - problem was the UTC. It's flashed with a nonfunctional firmware and DFU mode is actually disabled by default so documentation is incorrect. You can enable DFU mode via Boot button if you have an STlink. More here:
https://github.com/Esoterical/voron_canbus/tree/main/can_adapter/Makerbase%20UTC%201.0

tntclaus, 2024-05-23:
Oh,  thank you @Armadillo1 that really makes perfect sense! I'll share my experience too.

Because I had to use USB I can say that USB turned out to be not bad at all :-D My personal experience so far is that USB in 3d printing machines for inter-mcu comms should be preferred over CAN, unless there is a serious reason to use CAN bus.  Ofc try to avoid standard usb connectors (type A/B/C) — they are too fragile and may not last long.

closed 2024-05-23 by tntclaus